mmoskal/uf2-stm32f #1: How small could this firmware go?

mithro, 2018-10-03:
I was wondering how small you could make a UF2 compatible firmware? Could you fit it into a M0+ class device? Could you combined it with something like @xobs' https://github.com/xobs/grainuum stack?

xobs, 2018-10-15:
I have an example of Grainuum running on an M0+ bootloader at https://github.com/xobs/joyboot however it's using HID and not U2F.  If I were to redo `joyboot` (to actually produce the boards) I would use U2F now.

mithro, 2018-10-15:
By that I assume you mean DFU rather than U2F?

mithro, 2018-10-15:
Or UF2?

xobs, 2018-10-15:
...yes, sorry.  I hadn't heard of UF2 before, but I suppose the `joyboot` firmware could be easily adapted to UF2.

mmoskal, 2019-04-25:
sorry for late reply, here's one for STM32F1 (which is M3, but small) https://github.com/mmoskal/uf2-stm32f103; here's one for SAMD21 (which is M0+) https://github.com/Microsoft/uf2-samdx1

I even had it running on the atmega usb interface chip of uno.

For full list see here https://github.com/Microsoft/uf2#bootloaders

closed 2019-04-25 by mmoskal