mmoskal/uf2-stm32f #7: The uf2(makecode arcade) is too large and USB mass storage can't open

lualiliu, 2019-10-11:
Hello!
I'm using stm32f411CEU6 and this board I flash bootloader.bin(MVD12) in 0x08000000.I found a new USB mass storage named likes "Arcade-F4" but I can't open.

i try to flash .uf2(download from arcade.makecode.com),i found every .uf2 is greater than 512k, i can't flash.

thank.

lualiliu, 2019-10-11:
[image]

lualiliu, 2019-10-11:
i try to
`python .\uf2conv.py .\arcade-a.uf2`
and i get the flash.bin(287kb), and i flash in APP_LOAD_ADDRESS(0x08010000)
the screen is white, what address is right?Thank

mmoskal, 2019-10-11:
UF2 files are generally 2x the size of the actual binary. You should be able to just drop the file on the `ARCADE-F4` drive. Are you getting some messages from operating system? Did you try a different computer? You should just be able to open the drive.

Loading at 0x08010000 should also work fine.

Can you connect gdb and send me dump of all RAM when it displays the white screen? There's an area where some logs are so they might be useful.

lualiliu, 2019-10-12:
win7 is work fine,thank!

closed 2019-10-12 by lualiliu

mmoskal, 2019-10-12:
Which operating system did you try before?