mmoskal/uf2-stm32f #10: Cannot flash uf2 bootloader to mbed enabled board

yuanyanhui, 2020-01-28:
I have a stm32 nucleo board with mbed enabled. I tried to flash the uf2 bootloader, but the board still shows up as a mbed drive instead of a uf2 drive. It seems the mbed bootloader is protected from erase.

Is it possible to put the uf2 bootloader on a mbed board?

EmbeddedProject, 2020-02-04:
ARM Mbed is another solution. The Boards contains two chips on the board - the target (main) chip and the interface chip, which communicates via USB with the computer and via debug wires with the target. The interface chip implements a virtual file system (more on this below) which presents itself to the computer as a USB stick. The user compiles the program in his browser and receives a file in **Intel HEX format**. The user then pulls this file onto the USB stick. The interface chip recognizes the hex format, and (with some temporary storage) it flashes over the debug wires to the target chip. More specifically, the interface chip writes the files to be flashed and a short flash program into the RAM of the target and executes it. No special drivers need to be installed, since operating systems support USB sticks by default.

yuanyanhui, 2020-02-09:
Thanks a lot for your clear explanation. Yes, mbed has one extra chip handling communication between PC and MCU. It appears that the mbed bootloader also resides on that interface chip.

closed 2020-02-09 by yuanyanhui