mmoskal/uf2-stm32f #21: weACT 2.0 stm32f411ce unable to enumerate USB device, probably board pullups

charkster, 2020-12-21:
I'm using the F104c board configuration and flashing my WeACT Blackpill 2.0
Under Raspberry Pi OS and x86 Ubuntu I am not able to see a storage device attach when first connected.
I need to press the Boot0 and NRST pins (DFU mode) to be able to see a dmesg result.

[19730.784295] usb 1-5: device descriptor read/64, error -71
[19731.020284] usb 1-5: device descriptor read/64, error -71
[19731.256268] usb 1-5: new full-speed USB device number 8 using xhci_hcd
[19731.384229] usb 1-5: device descriptor read/64, error -71
[19731.620221] usb 1-5: device descriptor read/64, error -71
[19731.728372] usb usb1-port5: attempt power cycle
[19732.380247] usb 1-5: new full-speed USB device number 9 using xhci_hcd
[19732.380409] usb 1-5: Device not responding to setup address.
[19732.588435] usb 1-5: Device not responding to setup address.
[19732.796258] usb 1-5: device not accepting address 9, error -71
[19732.924262] usb 1-5: new full-speed USB device number 10 using xhci_hcd
[19732.924476] usb 1-5: Device not responding to setup address.
[19733.132300] usb 1-5: Device not responding to setup address.
[19733.340171] usb 1-5: device not accepting address 10, error -71
[19733.340241] usb usb1-port5: unable to enumerate USB device

I have tried the "#define USB_FORCE_DISCONNECT 1" in the board.h file, but I get the same message.

I am flashing to address 0x08000000, which is where I believe it starts execution from.
Here is my flash command:
sudo st-flash write bootloader.bin 0x08000000

Here is the STM32 wiki page for this board:
https://stm32-base.org/boards/STM32F411CEU6-WeAct-Black-Pill-V2.0.html

The wiki page does describe extra connections made to D+ and D-:
  # |  Function | Connected to
B7 |  D-           | PA11 via 10Ω (R9)
A6 |  D+          | PA12 via 10Ω (R7)
A7 |  D-           | PA11 via 10Ω (R9)
B6 |  D+          | PA12 via 10Ω (R7)

Should I pull these resistors off my board, or do they have a USB protection feature?
Any ideas would be appreciated.

mmoskal, 2020-12-22:
The resistors shouldn't make a difference.

If you're not trying to build a MakeCode Arcade device, I suggest you try https://github.com/adafruit/tinyuf2 instead of this repo.

charkster, 2020-12-22:
Thank you for the suggestion. I will use that project.

closed 2020-12-22 by charkster