mmoskal/uf2-stm32f #27: Build errors

aranis22, 2025-03-31:
Getting build errors when trying to run make:
C:/ProgramData/chocolatey/lib/make/tools/install/bin/make.exe checksubmodules
make[1]: Entering directory 'C:/Users/aranis/Desktop/uf2/uf2-stm32f'
(git submodule init)
Submodule 'libopencm3' (https://github.com/mmoskal/libopencm3.git) registered for path 'libopencm3'
Submodule 'uf2' (https://github.com/Microsoft/uf2) registered for path 'uf2'
(git submodule update)
Cloning into 'C:/Users/aranis/Desktop/uf2/uf2-stm32f/libopencm3'...
Cloning into 'C:/Users/aranis/Desktop/uf2/uf2-stm32f/uf2'...
Submodule path 'libopencm3': checked out '364ee72d0c05844c703c86e0bf4f407443a7370a'
Submodule path 'uf2': checked out '84444ddb9d2914edf7f6d9e89a7ce41b53d64d98'
(./Tools/check_submodules.sh)
Checked libopencm3 submodule, correct version found
make[1]: Leaving directory 'C:/Users/aranis/Desktop/uf2/uf2-stm32f'
C:/ProgramData/chocolatey/lib/make/tools/install/bin/make.exe -C libopencm3 -j10 lib
  GENHDR  include/libopencm3/efm32/g/irq.json
  GENHDR  include/libopencm3/efm32/gg/irq.json
  GENHDR  include/libopencm3/efm32/lg/irq.json
  GENHDR  include/libopencm3/lm3s/irq.json
  GENHDR  include/libopencm3/efm32/tg/irq.json
  GENHDR  include/libopencm3/lpc13xx/irq.json
  GENHDR  include/libopencm3/lpc17xx/irq.json
  GENHDR  include/libopencm3/lpc43xx/m0/irq.json
  GENHDR  include/libopencm3/sam/3a/irq.json
  GENHDR  include/libopencm3/lpc43xx/m4/irq.json
  GENHDR  include/libopencm3/sam/3n/irq.json
  GENHDR  include/libopencm3/sam/3s/irq.json
  GENHDR  include/libopencm3/sam/3u/irq.json
  GENHDR  include/libopencm3/sam/3x/irq.json
  GENHDR  include/libopencm3/sam/d/irq.json
  GENHDR  include/libopencm3/stm32/f0/irq.json
  GENHDR  include/libopencm3/stm32/f1/irq.json
  GENHDR  include/libopencm3/stm32/f2/irq.json
  GENHDR  include/libopencm3/stm32/f3/irq.json
  GENHDR  include/libopencm3/stm32/f4/irq.json
  GENHDR  include/libopencm3/stm32/f7/irq.json
  GENHDR  include/libopencm3/stm32/l0/irq.json
  GENHDR  include/libopencm3/stm32/l1/irq.json
  GENHDR  include/libopencm3/stm32/l4/irq.json
  GENHDR  include/libopencm3/vf6xx/irq.json
  BUILD   lib/stm32/f1
  BUILD   lib/stm32/f0
  BUILD   lib/stm32/f2
  BUILD   lib/stm32/f3
  BUILD   lib/stm32/f4
  BUILD   lib/stm32/f7
  BUILD   lib/stm32/l1
  BUILD   lib/lpc13xx
  BUILD   lib/stm32/l0
  BUILD   lib/stm32/l4
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
make[1]: *** [Makefile:63: lib/stm32/f1] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
make[1]: *** [Makefile:63: lib/stm32/f3] Error 2
make[1]: *** [Makefile:63: lib/lpc13xx] Error 2
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
make[1]: *** [Makefile:63: lib/stm32/f7] Error 2
make[1]: *** [Makefile:63: lib/stm32/f2] Error 2
make[1]: *** [Makefile:63: lib/stm32/l1] Error 2
make[1]: *** [Makefile:63: lib/stm32/f4] Error 2
make[1]: *** [Makefile:63: lib/stm32/f0] Error 2
make[1]: *** [Makefile:63: lib/stm32/l0] Error 2
make[1]: *** [Makefile:63: lib/stm32/l4] Error 2
make: *** [Makefile:146: libopencm3/include/libopencm3/stm32/f4/nvic.h] Error 2

HomeAssistantTycoon, 2025-08-14:
I’m having the same issue

HomeAssistantTycoon, 2025-12-15:
I managed to fix the issue at least in GitHub actions it’ll take around three hours to build but it should now work: https://github.com/HomeAssistantTycoon/BetterMeowbit